ptrkrysik/gr-gsm #601: Gr-gsm bladerf 2.0 A5 stuck in loop when scan (Ubuntu 20.04)

KimlongSeng, 2022-04-06:
Ubuntu 20.04
Gnuradio 3.8 : sudo apt install gnuradio
Gr-gsm install :

git clone -b maint-3.8 https://github.com/velichkov/gr-gsm.git
cd gr-gsm
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig
export PYTHONPATH=/usr/local/lib/python3/dist-packages/:$PYTHONPATH

Bladerf infor
: bladeRF-cli -e info -e version

INFO @ host/libraries/libbladeRF/src/helpers/version.c:79] Firmware version (v2.4.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:103] FPGA version (v0.14.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.

  Board:                    Nuand bladeRF 2.0 (bladerf2)
  Serial #:                 09d2840b349b4c90bb5d740e892111a8
  VCTCXO DAC calibration:   0x1c71
  FPGA size:                Unknown
  FPGA loaded:              yes
  Flash size:               32 Mbit (assumed)
  USB bus:                  2
  USB address:              3
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0

  bladeRF-cli version:        1.8.0-0.2019.07-4build1
  libbladeRF version:         2.2.1-0.2019.07-4build1

  Firmware version:           2.4.0-git-a3d5c55f
  FPGA version:               0.14.0 (configured by USB host)

- Issue 
$ grgsm-scanner -b PCS1900 --args=bladerf -d

output:

Args= bladerf
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string ''
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:79] Firmware version (v2.4.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:103] FPGA version (v0.14.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[bladeRF source] Device: Nuand bladeRF 2.0 Serial # 09d2...11a8 FW v2.4.0 FPGA v0.14.0
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
[bladeRF source] set_dc_offset_mode: Automatic DC correction mode is not implemented.
[bladeRF source] start: DEBUG: starting source
[bladeRF source] stop: DEBUG: stopping source
Args= bladerf
.................................................................................[Repeat] ................................................................................................

any help will be appreciate.

velichkov, 2022-04-07:
Hi @KimlongSeng,

> Gr-gsm bladerf 2.0 A5 stuck in loop when scan

It is not stuck, it scans the selected band in 2MHz chunks and on every iteration it creates a new instance and prints the messages you are seeing.

Make sure you have a suitable antenna attached to your SDR device and in case you are inside of a big building try moving near a window or outside if possible.

> INFO @ host/libraries/libbladeRF/src/helpers/version.c:79] Firmware version (v2.4.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
> [INFO @ [...]

I don't have any experience with bladeRF but you probably need to resolve those by either downgrading the firmware and FPGA or upgrading libbladeRF and then rebuild/reinstall gr-osmosdr and gr-gsm.

KimlongSeng, 2022-04-07:
Thank you for the response @velichkov. In that case, I will close this issue. Have a great day.

closed 2022-04-07 by KimlongSeng