ptrkrysik/gr-gsm #603: cmake .. Configuring incomplete, errors occurred!

jihatbabuislam, 2022-04-28:
**please help me**   Configuring incomplete, errors occurred!

┌──(root㉿kali)-[~/gr-gsm/build]
└─# cmake ../          
-- Build type not specified: defaulting to release.
-- Using GMP.
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.10.so (found suitable exact version "3.10.4") 
-- Checking for module 'sndfile'
--   No package 'sndfile' found
-- Could NOT find SNDFILE (missing: SNDFILE_LIBRARIES SNDFILE_INCLUDE_DIRS) 
-- Extracting version information from git describe...
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):                                                           
  The package name passed to `find_package_handle_standard_args` (PkgConfig)     
  does not match the name of the calling package (Libosmocore).  This can        
  lead to problems in calling code that expects `find_package` result            
  variables (e.g., `_FOUND`) to follow a certain pattern.                        
Call Stack (most recent call first):                                             
  /usr/share/cmake-3.23/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)                                                                        
  cmake/Modules/FindLibosmocore.cmake:2 (INCLUDE)                                
  CMakeLists.txt:132 (find_package)                                              
This warning is for project developers.  Use -Wno-dev to suppress it.            
                                                                                 
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):                                                           
  The package name passed to `find_package_handle_standard_args`                 
  (libosmocore) does not match the name of the calling package (Libosmocore).    
  This can lead to problems in calling code that expects `find_package`          
  result variables (e.g., `_FOUND`) to follow a certain pattern.                 
Call Stack (most recent call first):                                             
  cmake/Modules/FindLibosmocore.cmake:32 (find_package_handle_standard_args)     
  CMakeLists.txt:132 (find_package)                                              
This warning is for project developers.  Use -Wno-dev to suppress it.            
                                                                                 
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):                                                           
  The package name passed to `find_package_handle_standard_args` (PkgConfig)     
  does not match the name of the calling package (Libosmocodec).  This can       
  lead to problems in calling code that expects `find_package` result            
  variables (e.g., `_FOUND`) to follow a certain pattern.                        
Call Stack (most recent call first):                                             
  /usr/share/cmake-3.23/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)                                                                        
  cmake/Modules/FindLibosmocodec.cmake:1 (INCLUDE)                               
  CMakeLists.txt:133 (find_package)                                              
This warning is for project developers.  Use -Wno-dev to suppress it.            
                                                                                 
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):                                                           
  The package name passed to `find_package_handle_standard_args`                 
  (libosmocodec) does not match the name of the calling package                  
  (Libosmocodec).  This can lead to problems in calling code that expects        
  `find_package` result variables (e.g., `_FOUND`) to follow a certain           
  pattern.                                                                       
Call Stack (most recent call first):                                             
  cmake/Modules/FindLibosmocodec.cmake:31 (find_package_handle_standard_args)    
  CMakeLists.txt:133 (find_package)                                              
This warning is for project developers.  Use -Wno-dev to suppress it.            
                                                                                 
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):                                                           
  The package name passed to `find_package_handle_standard_args`                 
  (libosmocoding) does not match the name of the calling package                 
  (Libosmocoding).  This can lead to problems in calling code that expects       
  `find_package` result variables (e.g., `_FOUND`) to follow a certain           
  pattern.                                                                       
Call Stack (most recent call first):                                             
  cmake/Modules/FindLibosmocoding.cmake:31 (find_package_handle_standard_args)   
  CMakeLists.txt:134 (find_package)                                              
This warning is for project developers.  Use -Wno-dev to suppress it.            
                                                                                 
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):                                                           
  The package name passed to `find_package_handle_standard_args` (PkgConfig)     
  does not match the name of the calling package (Libosmogsm).  This can lead    
  to problems in calling code that expects `find_package` result variables       
  (e.g., `_FOUND`) to follow a certain pattern.                                  
Call Stack (most recent call first):                                             
  /usr/share/cmake-3.23/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)                                                                        
  cmake/Modules/FindLibosmogsm.cmake:1 (INCLUDE)                                 
  CMakeLists.txt:135 (find_package)                                              
This warning is for project developers.  Use -Wno-dev to suppress it.            
                                                                                 
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):                                                           
  The package name passed to `find_package_handle_standard_args` (libosmogsm)    
  does not match the name of the calling package (Libosmogsm).  This can lead    
  to problems in calling code that expects `find_package` result variables       
  (e.g., `_FOUND`) to follow a certain pattern.                                  
Call Stack (most recent call first):                                             
  cmake/Modules/FindLibosmogsm.cmake:31 (find_package_handle_standard_args)      
  CMakeLists.txt:135 (find_package)                                              
This warning is for project developers.  Use -Wno-dev to suppress it.            
                                                                                 
-- Loading build date Thu, 28 Apr 2022 16:56:14 into constants...
-- Loading version v0.42.2-99-g2de47e28 into constants...
-- Using install prefix: /usr/local
-- Building for version: v0.42.2-99-g2de47e28 / 1.0.0git
-- No C++ unit tests... skipping
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.10.so (found version "3.10.4") 
CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find requested file:                                         
                                                                                 
    GrSwig                                                                       
                                                                                 
                                                                                 
CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".                                          
                                                                                 
                                                                                 
-- Configuring incomplete, errors occurred!
See also "/root/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/root/gr-gsm/build/CMakeFiles/CMakeError.log".
                                                                                 
┌──(root㉿kali)-[~/gr-gsm/build]
└─#

darkhacknet, 2022-05-21:
Same error here ! :/

MartinxMax, 2022-06-22:
Maybe you can use this -->#apt-get install libsndfile1-dev

jihatbabuislam, 2022-07-20:
yes i install it apt-get install libsndfile1-dev  done

but 

show 

cmake ..
-- The CXX compiler identification is GNU 11.3.0
-- The C compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'gmp'
--   Found gmp, version 6.2.1
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Using GMP.
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.74.0") found components: date_time program_options system regex thread unit_test_framework 
-- Found Volk: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.10.4") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.10.so (found suitable exact version "3.10.4") 
-- Checking for module 'sndfile'
--   Found sndfile, version 1.0.31
-- Found SNDFILE: /usr/lib/x86_64-linux-gnu/libsndfile.so  
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.8
-- Found FFTW3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found Git: /usr/bin/git  
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.1") found components: doxygen dot 
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (Libosmocore).  This can
  lead to problems in calling code that expects `find_package` result
  variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.23/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  cmake/Modules/FindLibosmocore.cmake:2 (INCLUDE)
  CMakeLists.txt:132 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'libosmocore'
--   Found libosmocore, version 1.6.0
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args`
  (libosmocore) does not match the name of the calling package (Libosmocore).
  This can lead to problems in calling code that expects `find_package`
  result variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Modules/FindLibosmocore.cmake:32 (find_package_handle_standard_args)
  CMakeLists.txt:132 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
[... 61 lines cut ...]
  CMakeLists.txt:135 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found libosmogsm: /usr/lib/x86_64-linux-gnu/libosmogsm.so  
-- Loading build date Wed, 20 Jul 2022 12:07:37 into constants...
-- Loading version v1.0-compat-xxx-xunknown into constants...
-- Using install prefix: /usr/local
-- Building for version: v1.0-compat-xxx-xunknown / 1.0.0git
-- No C++ unit tests... skipping
CMake Warning at docs/CMakeLists.txt:54 (message):
  rst2man from python-docutils is required to build man pages

-- Found SWIG: /usr/bin/swig4.0 (found version "4.0.2")  
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.10.so (found version "3.10.4") 
CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find requested file:

    GrSwig

CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

-- Configuring incomplete, errors occurred!
See also "/home/babu/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/home/babu/gr-gsm/build/CMakeFiles/CMakeError.log".

jihatbabuislam, 2022-07-20:
please help sir

Mhagne, 2022-08-27:
CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find requested file:                                                            
                                                                                                    
    GrSwig                                                                                          
                                                                                                    
                                                                                                    
CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

greyhat94, 2022-09-16:
I'm experiencing the same issues. As for the tools required, I have installed them with the command "sudo apt install toolname", thinking that the installation included the headers.