ptrkrysik/gr-gsm #605: how to solve grgsm_livemon.py

ayao999, 2022-06-19:
sudo grcc grgsm_livemon.grc

grgsm_livemon.grc:1:1:FATAL:PARSER:ERR_DOCUMENT_EMPTY: Start tag expected, '<' not found
Error during file compilation

ayao999, 2022-06-24:
I am referring to this article https://osmocom.org/projects/gr-gsm/wiki/Installation's Installation from packages on Debian Testing and Ubuntu 18.04+

ayao999, 2022-06-24:
I tried to build grgsm_livemon at ubuntu18.04, but couldn't rebuild it, as this article says, as the Article says, As of gr-gsm's version 0.41.2 grgsm_livemon application doesn't work because of change of PyQT version in current Debian Testing and Ubuntu 18.04. In order to use this program you can generate new version out of "grgsm_livemon.grc"

ayao999, 2022-06-24:
I hope you can understand

velichkov, 2022-06-24:
> I am referring to this article https://osmocom.org/projects/gr-gsm/wiki/Installation's Installation from packages on Debian Testing and Ubuntu 18.04+

OK, this means you have gnuradio 3.7.11-10 

> I tried to build grgsm_livemon at ubuntu18.04, but couldn't rebuild it, as this article says, as the Article says, As of gr-gsm's version 0.41.2 grgsm_livemon application doesn't work because of [...]

Where did you got grgsm_livemon.grc from? Provide the full output of `head -10 grgsm_livemon.grc`

DroidKali, 2022-07-03:
Same issue for me.

DroidKali, 2022-07-03:
> OK, this means you have gnuradio 3.7.11-10
>
>
> Where did you got grgsm_livemon.grc from? Provide the full output of `head -10 grgsm_livemon.grc`

The grgsm_livemon.grc is from https://gitea.osmocom.org/sdr/gr-gsm/raw/branch/master/apps/grgsm_livemon.grc

The output of `head -10 grgsm_livemon.grc`
```
options:
  parameters:
    author: Piotr Krysik
    category: Custom
    cmake_opt: ''
    comment: ''
    copyright: ''
    description: 'Interactive monitor of a single C0 channel with analysis performed
      by Wireshark (command to run wireshark: sudo wireshark -k -f udp -Y gsmtap -i
      lo)'
    gen_cmake: 'On'
    gen_linking: dynamic
```

ayao999, 2022-07-06:
Thank you!I already save it.

 
Hi @DroidKali,
  

 
 
  
Currently the master branch contains the version for gnruadio 3.8. In gnuradio 3.8 the format of grc files was changed from xml to yaml but the file extension remained the same - .grc
 
In Ubuntu 18.04 the gnuradio is 3.7.11-10, this version does not support yaml format and because of this you are getting this error.
  
  
You need to download an older version of grgsm_livemon.grc from the v0.41.1 tag.
 
https://gitea.osmocom.org/sdr/gr-gsm/raw/tag/v0.41.1/apps/grgsm_livemon.grc

DroidKali, 2022-07-06:
> Hi @DroidKali,
>
>
> Currently the `master` branch contains the version for gnruadio 3.8. In gnuradio 3.8 the format of grc files was changed from `xml` to `yaml` but the file extension remained the [...]

It works！Thanks for your help！

ptrkrysik, 2026-11-09:
The .grc files are now YAML for GNU Radio 3.8+ and the issue was caused by using them with GNU Radio 3.7; solved in the thread, closing.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik