ptrkrysik/gr-gsm #606: pybombs installation error

310564734, 2022-06-24:
When I entered this command, an error occured

`rty@rty-desktop:~/gnuradio-prefix$ pybombs install gnuradio
[INFO] Prefix Python version is: 3.6.9
[INFO] PyBOMBS Version 2.3.5
[INFO] Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: gnuradio
[WARNING] Build dir already exists: /home/rty/gnuradio-prefix/src/gnuradio/build
Configuring: (100%) [=======================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Build date set to Fri, 24 Jun 2022 03:17:45Z.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- Extracting version information from git describe...
-- 
-- Configuring testing-support support...
--   Dependency Boost_UNIT_TEST_FRAMEWORK_FOUND = 1
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
CMake Warning at cmake/Modules/GrCompilerSettings.cmake:26 (MESSAGE):
  

  The compiler selected to build GNU Radio (GCC version 8.4.0 : /usr/bin/c++)
  is older than that officially supported (9.3.0 minimum).  This build may or
  not work.  We highly recommend using a more recent GCC version.
Call Stack (most recent call first):
  CMakeLists.txt:99 (include)

-- Compiler Version: cc (Ubuntu/Linaro 8.4.0-1ubuntu1~18.04) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG -I/usr/local/cuda/include -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- User set python executable /usr/bin/python3.6
-- Python checking for packaging - found
-- Python checking for numpy - found
-- Python checking for pygccxml - found
-- Found pybind11: /usr/local/include (found version "2.10.0dev1")
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency PACKAGING_FOUND = TRUE
--   Dependency pybind11_FOUND = 1
--   Dependency NUMPY_FOUND = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- 
-- Configuring VOLK support...
--   Found VOLK:
--   * Version: 2.5.1
--   * Libraries: Volk::volk
--   * Includes: /home/rty/gnuradio-prefix/include
-- Not using additional GNU Radio native architecture optimizations.
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Using GMP.
-- 
-- Configuring post-install support...
--   Enabling post-install support.
--   Override with -DENABLE_POSTINSTALL=ON/OFF
-- Could NOT find MathJax2 (missing: MATHJAX2_JS_PATH) 
-- 
-- Configuring doxygen support...
--   Dependency DOXYGEN_FOUND = YES
--   Enabling doxygen support.
--   Override with -DENABLE_DOXYGEN=ON/OFF
-- 
-- Configuring man-pages support...
--   Enabling man-pages support.
--   Override with -DENABLE_MANPAGES=ON/OFF
CMake Warning at docs/doxygen/CMakeLists.txt:24 (message):
  MathJax 2 not found, HTML equations might not be properly rendered.

-- Checking for module 'gmp'
--   No package 'gmp' found
-- Using GMP.
-- 
-- Configuring gnuradio-runtime support...
--   Dependency Boost_FOUND = TRUE
--   Dependency PYTHONINTERP_FOUND = TRUE
--   Dependency MPLIB_FOUND = TRUE
--   Dependency spdlog_FOUND = 1
--   Dependency pybind11_FOUND = 1
--   Dependency Volk_FOUND = 1
--   Enabling gnuradio-runtime support.
--   Override with -DENABLE_GNURADIO_RUNTIME=ON/OFF
-- 
-- Configuring common-precompiled-headers support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency GR_CMAKE_HAS_PCH = 3.22.5
--   Enabling common-precompiled-headers support.
--   Override with -DENABLE_COMMON_PCH=ON/OFF
-- 
-- Configuring gr-ctrlport support...
--   Dependency Boost_FOUND = TRUE
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Enabling gr-ctrlport support.
--   Override with -DENABLE_GR_CTRLPORT=ON/OFF
-- Checking for module 'thrift'
--   No package 'thrift' found
-- thrift looking for version 0.13
-- Binary 'thrift' not found.
-- Checking for module 'libunwind'
--   No package 'libunwind' found
-- Loading build date Fri, 24 Jun 2022 03:17:45Z into constants...
-- Loading version 3.10.3.0 into constants...
-- TRY_SHM_VMCIRCBUF set to ON.
-- 
-- Python checking for PyYAML >= 3.11 - found
-- Python checking for mako >= 1.1.0 - not found
-- Python checking for pygobject >= 2.28.6 - found
-- Python checking for Gtk (GI) >= 3.10.8 - found
-- Python checking for Cairo (GI) >= 1.0 - found
-- Python checking for PangoCairo (GI) >= 1.0 - found
-- Python checking for numpy - found
-- Python checking for jsonschema - found
-- 
-- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency PYYAML_FOUND = TRUE
--   Dependency MAKO_FOUND = FALSE
--   Dependency PYGI_FOUND = TRUE
--   Dependency GTK_GI_FOUND = TRUE
--   Dependency CAIRO_GI_FOUND = TRUE
--   Dependency PANGOCAIRO_GI_FOUND = TRUE
--   Dependency NUMPY_FOUND = TRUE
CMake Error at cmake/Modules/GrComponent.cmake:63 (message):
  user force-enabled gnuradio-companion but configuration checked failed
Call Stack (most recent call first):
  grc/CMakeLists.txt:81 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/rty/gnuradio-prefix/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/rty/gnuradio-prefix/src/gnuradio/build/CMakeFiles/CMakeError.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gnuradio:
Configuration failed
[ERROR] Error installing package gnuradio. Aborting.

`

310564734, 2022-06-26:
@velichkov ,Thanks, I tried to install the latest Mako, and it did work. But another error occurred while building.

`rty@rty-desktop:~/gnuradio-prefix$ pybombs install gnuradio
[INFO] Prefix Python version is: 3.6.9
[INFO] PyBOMBS Version 2.3.5
[INFO] Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: gnuradio
[WARNING] Build dir already exists: /home/rty/gnuradio-prefix/src/gnuradio/build
Building:    (100%) [=========================================================]
[  0%] Built target doxygen_target
[  0%] Built target gnuradio_docstrings
Consolidate compiler generated dependencies of target common-precompiled-headers
[  0%] Building CXX object gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx.gch
In file included from /home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/custom_lock.h:15,
                 from /home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/buffer.h:15,
                 from /home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/buffer_double_mapped.h:15,
                 from /home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/io_signature.h:17,
                 from /home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/basic_block.h:15,
                 from /home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/hier_block2.h:15,
                 from /home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/top_block.h:15,
                 from /home/rty/gnuradio-prefix/src/gnuradio/build/gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx:5,
                 from <command-line>:
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:140:37: error: ‘string_view_t’ in namespace ‘spdlog’ does not name a type
     inline void trace(const spdlog::string_view_t& msg, const Args&... args)
                                     ^~~~~~~~~~~~~
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:147:37: error: ‘string_view_t’ in namespace ‘spdlog’ does not name a type
     inline void debug(const spdlog::string_view_t& msg, const Args&... args)
                                     ^~~~~~~~~~~~~
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:154:36: error: ‘string_view_t’ in namespace ‘spdlog’ does not name a type
     inline void info(const spdlog::string_view_t& msg, const Args&... args)
                                    ^~~~~~~~~~~~~
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:161:38: error: ‘string_view_t’ in namespace ‘spdlog’ does not name a type
     inline void notice(const spdlog::string_view_t& msg, const Args&... args)
                                      ^~~~~~~~~~~~~
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:168:36: error: ‘string_view_t’ in namespace ‘spdlog’ does not name a type
     inline void warn(const spdlog::string_view_t& msg, const Args&... args)
                                    ^~~~~~~~~~~~~
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:175:37: error: ‘string_view_t’ in namespace ‘spdlog’ does not name a type
     inline void error(const spdlog::string_view_t& msg, const Args&... args)
                                     ^~~~~~~~~~~~~
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:182:36: error: ‘string_view_t’ in namespace ‘spdlog’ does not name a type
     inline void crit(const spdlog::string_view_t& msg, const Args&... args)
                                    ^~~~~~~~~~~~~
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:189:37: error: ‘string_view_t’ in namespace ‘spdlog’ does not name a type
     inline void alert(const spdlog::string_view_t& msg, const Args&... args)
                                     ^~~~~~~~~~~~~
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:196:37: error: ‘string_view_t’ in namespace ‘spdlog’ does not name a type
     inline void fatal(const spdlog::string_view_t& msg, const Args&... args)
                                     ^~~~~~~~~~~~~
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:203:37: error: ‘string_view_t’ in namespace ‘spdlog’ does not name a type
     inline void emerg(const spdlog::string_view_t& msg, const Args&... args)
                                     ^~~~~~~~~~~~~
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:278:13: error: ‘formatter’ is not a class template
 struct fmt::formatter<boost::format> : formatter<string_view> {
             ^~~~~~~~~
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:278:38: error: qualified name does not name a class before ‘:’ token
 struct fmt::formatter<boost::format> : formatter<string_view> {
                                      ^
/home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:278:38: error: expected ‘{’ before ‘:’ token
[... cut ...]
make[2]: *** [gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx.gch] Error 1
CMakeFiles/Makefile2:3980: recipe for target 'gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/all' failed
make[1]: *** [gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package gnuradio:
Build failed.
[ERROR] Error installing package gnuradio. Aborting.

`

velichkov, 2022-06-26:
> -- Dependency spdlog_FOUND = 1

> /home/rty/gnuradio-prefix/src/gnuradio/gnuradio-runtime/include/gnuradio/logger.h:140:37: error: ‘string_view_t’ in namespace ‘spdlog’ does not name a type
inline void trace(const spdlog::string_view_t& msg, const Args&... args)

As you are having problems installing `gnuradio` my advice is to seek help in the gnuradio chat or mailing list

* https://wiki.gnuradio.org/index.php/Chat
* https://wiki.gnuradio.org/index.php/MailingLists

ptrkrysik, 2026-11-09:
This is a failure while building GNU Radio with PyBOMBS, not in gr-gsm. gr-gsm now builds against the GNU Radio packaged by the distribution, please use that.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.