ptrkrysik/gr-gsm #607: I hope you guys can solve grsm's make step

ayao999, 2022-06-28:
I tried to install and compile grrgsm in ubuntu16.04, where grrgsm comes from your github latest link https://github.com/ptrkrysik/gr-gsm,gnuraio version is 3.8, use git clone to manually compile the source code, and there is an error in the make link as described below

ayao999, 2022-06-28:
[  1%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/control_channels_decoder_impl.cc.o
/home/ayao999/gr-gsm/lib/decoding/control_channels_decoder_impl.cc: In constructor ‘gr::gsm::control_channels_decoder_impl::control_channels_decoder_impl()’:
/home/ayao999/gr-gsm/lib/decoding/control_channels_decoder_impl.cc:73:107: error: ‘boost::placeholders’ has not been declared
 ), boost::bind(&control_channels_decoder_impl::decode, this, boost::placeholder
                                                                     ^
lib/CMakeFiles/grgsm.dir/build.make:230: recipe for target 'lib/CMakeFiles/grgsm.dir/decoding/control_channels_decoder_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/grgsm.dir/decoding/control_channels_decoder_impl.cc.o] Error 1
CMakeFiles/Makefile2:347: recipe for target 'lib/CMakeFiles/grgsm.dir/all' failed
make[1]: *** [lib/CMakeFiles/grgsm.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
ayao999@ayao999-virtual-machine:~/gr-gsm/build$

ayao999, 2022-06-28:
ayao999@ayao999-virtual-machine:~/gr-gsm/build$ make
[  1%] Building CXX object lib/CMakeFiles/grgsm.dir/decoding/control_channels_decoder_impl.cc.o
/home/ayao999/gr-gsm/lib/decoding/control_channels_decoder_impl.cc: In constructor ‘gr::gsm::control_channels_decoder_impl::control_channels_decoder_impl()’:
/home/ayao999/gr-gsm/lib/decoding/control_channels_decoder_impl.cc:73:107: error: ‘boost::placeholders’ has not been declared
 ), boost::bind(&control_channels_decoder_impl::decode, this, boost::placeholder
                                                                     ^
lib/CMakeFiles/grgsm.dir/build.make:230: recipe for target 'lib/CMakeFiles/grgsm.dir/decoding/control_channels_decoder_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/grgsm.dir/decoding/control_channels_decoder_impl.cc.o] Error 1
CMakeFiles/Makefile2:347: recipe for target 'lib/CMakeFiles/grgsm.dir/all' failed
make[1]: *** [lib/CMakeFiles/grgsm.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
ayao999@ayao999-virtual-machine:~/gr-gsm/build$

ayao999, 2022-06-28:
Please check it out

ptrkrysik, 2026-11-09:
This was reported against the old GNU Radio 3.7/3.8 (SWIG) code base. The module has been ported to GNU Radio 3.10 and no longer uses SWIG or the old build, so please retry with the new release and open a new issue with the full log if the problem persists.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik