ptrkrysik/gr-gsm #608: gr gsm (grgsm_livemon generate Error)'''

pixkike, 2022-07-15:
Hi Sir 
I Running kali 2022 5.15.0-kali3-amd64,
I run make in grgsm/build  and i stuck in this point 
`make                                                                  
Consolidate compiler generated dependencies of target grgsm
[ 64%] Built target grgsm
[ 67%] Built target _grgsm_swig_doc_tag
[ 69%] Built target grgsm_swig_swig_doc
[ 70%] Built target grgsm_swig_swig_compilation
[ 72%] Built target grgsm_swig
[ 73%] Generating grgsm_livemon
Block with id "gsm_bcch_ccch_demapper" loaded from
  /usr/local/gr-gsm/grc/demapping/gsm_bcch_ccch_demapper.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_bcch_ccch_demapper.block.yml
Block with id "gsm_bcch_ccch_sdcch4_demapper" loaded from
  /usr/local/gr-gsm/grc/demapping/gsm_bcch_ccch_sdcch4_demapper.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_bcch_ccch_sdcch4_demapper.block.yml
Block with id "gsm_burst_file_sink" loaded from
  /usr/local/gr-gsm/grc/misc_utils/gsm_burst_file_sink.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_burst_file_sink.block.yml
Block with id "gsm_burst_file_source" loaded from
  /usr/local/gr-gsm/grc/misc_utils/gsm_burst_file_source.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_burst_file_source.block.yml
Block with id "gsm_burst_fnr_filter" loaded from
  /usr/local/gr-gsm/grc/flow_control/gsm_burst_fnr_filter.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_burst_fnr_filter.block.yml
Block with id "gsm_burst_sdcch_subslot_filter" loaded from
  /usr/local/gr-gsm/grc/flow_control/gsm_burst_sdcch_subslot_filter.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_burst_sdcch_subslot_filter.block.yml
Block with id "gsm_burst_sdcch_subslot_splitter" loaded from
  /usr/local/gr-gsm/grc/flow_control/gsm_burst_sdcch_subslot_splitter.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_burst_sdcch_subslot_splitter.block.yml
Block with id "gsm_burst_timeslot_filter" loaded from
  /usr/local/gr-gsm/grc/flow_control/gsm_burst_timeslot_filter.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_burst_timeslot_filter.block.yml
Block with id "gsm_burst_timeslot_splitter" loaded from
  /usr/local/gr-gsm/grc/flow_control/gsm_burst_timeslot_splitter.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_burst_timeslot_splitter.block.yml
Block with id "gsm_burst_to_fn_time" loaded from
  /usr/local/gr-gsm/grc/misc_utils/gsm_burst_to_fn_time.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_burst_to_fn_time.block.yml
Block with id "gsm_burst_type_filter" loaded from
  /usr/local/gr-gsm/grc/flow_control/gsm_burst_type_filter.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_burst_type_filter.block.yml
Block with id "gsm_bursts_printer" loaded from
  /usr/local/gr-gsm/grc/misc_utils/gsm_bursts_printer.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_bursts_printer.block.yml
Block with id "gsm_clock_offset_control" loaded from
  /usr/local/gr-gsm/grc/receiver/gsm_clock_offset_control.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_clock_offset_control.block.yml
Block with id "gsm_clock_offset_corrector_tagged" loaded from
  /usr/local/gr-gsm/grc/misc_utils/gsm_clock_offset_corrector_tagged.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_clock_offset_corrector_tagged.block.yml
Block with id "gsm_collect_system_info" loaded from
  /usr/local/gr-gsm/grc/misc_utils/gsm_collect_system_info.block.yml
overwritten by
[... 129 lines cut ...]
Block with id "gsm_universal_ctrl_chans_demapper" loaded from
  /usr/local/gr-gsm/grc/demapping/gsm_universal_ctrl_chans_demapper.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_universal_ctrl_chans_demapper.block.yml
Block with id "gsm_uplink_downlink_splitter" loaded from
  /usr/local/gr-gsm/grc/flow_control/gsm_uplink_downlink_splitter.block.yml
overwritten by
  /usr/local/share/gnuradio/grc/blocks/gsm_uplink_downlink_splitter.block.yml
<<< Welcome to GNU Radio Companion Compiler 3.10.3.0 >>>

Block paths:
        /root/.grc_gnuradio
        /usr/local/gr-gsm/grc
        /usr/local/share/gnuradio/grc/blocks

>>> Loading: /usr/local/gr-gsm/apps/grgsm_livemon.grc
>>> Generating: /usr/local/gr-gsm/build/apps/grgsm_livemon.py
>>> Generate Error: /usr/local/gr-gsm/apps/grgsm_livemon.grc: 'catch_exceptions'
Compilation error
Segmentation fault
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:97: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:862: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:146: all] Error 2`
Please Tell Me How To Fix This .... 
Sorry For My Bad writing ......

pixkike, 2022-07-16:
This is my output of cmake VERBOSE=1 @log.txt

velichkov, 2022-07-18:
Hi @pixkike,

> I run make in grgsm/build and i stuck in this point

Which gr-gsm version and from which repository/branch?

pixkike, 2022-07-18:
Thank you so much For Replying me sir  grgsm  is not installed yet I download grgsm from this link https://git.osmocom.org/gr-gsm ,
and install gnuradio 3.8 from this https://github.com/ptrkrysik/gr-gsm/issues/597#issuecomment-1040038428 ,
this is my kali repo deb http://http.kali.org/kali kali-rolling main contrib non-free

velichkov, 2022-07-19:
>  i haven't completely uninstall gnuradio 3.10 so i delete all of gnuradio 3.10

How did you deleted it? 

> reinstall 3.8

How did you reinstall 3.8, did you built it  from source? When did you reinstall it - before or after deleting 3.10?

>```
> >>> Loading: /usr/local/gr-gsm/apps/grgsm_livemon.grc
> Traceback (most recent call last):
> File "/usr/local/lib/python3.10/dist-packages/osmosdr/__init__.py", line 17, in <module>
>  from .osmosdr_python import *
> ImportError: libgnuradio-funcube.so.1.0.0: cannot open shared object file: No such file or directory
>```

Looks like your `gr-osmosdr` was built with `libgnuradio-funcube` support? Did you remove `libgnuradio-funcube.so.1.0.0`? When did you built `gr-smosdr` - before or after deleting 3.10?

You need to uninstall `gr-osmosdr` and do a clean rebuild without  `libgnuradio-funcube` support.

P.S.
You can find a version of `gr-gsm` that supports gnuradio 3.10 in [bkerler's fork (maint-3.10 branch)](https://github.com/bkerler/gr-gsm/tree/maint-3.10)

pixkike, 2022-07-19:
> How did you deleted it?

at first I have 2 gnuradio versions in my system 
So 1st of all I delete my gnuradio 3.8 and
2nd  i uninstall with "synaptic package manager" gnuradio and all related dependencies and then i type 
"find /usr/local -name gnuradio" and remove after showing gnuradio location all

>How did you reinstall 3.8, did you built it from source? When did you reinstall it - before or after deleting 3.10?

After deleting all dependencies I reinstalll 3.8 from the source ,and yep after deleting 3.10 

> Did you remove libgnuradio-funcube.so.1.0.0?

Yep I did This But Right now I reinstall this with " apt install libgnuradio-osmosdr0.2.0 "

>You need to uninstall gr-osmosdr and do a clean rebuild without libgnuradio-funcube support.

OHH can I install any versions gr-smosdr

>P.S.
You can find a version of gr-gsm that supports gnuradio 3.10 in [bkerler's fork (maint-3.10 branch)](https://github.com/bkerler/gr-gsm/tree/maint-3.10)
 v

Ohh really ? Can I install grgsm with gnuradio 3.10 updated versions ?

velichkov, 2022-07-19:
> > Did you remove libgnuradio-funcube.so.1.0.0?
> 
> Yep I did This But Right now I reinstall this with " apt install libgnuradio-osmosdr0.2.0 "

In your case you should not install gr-osmosdr from apt repository. Once you install gnuradio from source you must install all other gnuradio's OOT modules (like osmosdr, gr-gsm .... ) from source as well. The versions from the apt repo were built for a different gnuradio version and after installing them you'll have two gnuradio versions at the same time.

> > You need to uninstall gr-osmosdr and do a clean rebuild without libgnuradio-funcube support.
> 

> OHH can I install any versions gr-smosdr

No, not any version. You need to install the version for gnuradio 3.8 (ususally from a branch with 3.8 in its name)

> > P.S.
> > You can find a version of gr-gsm that supports gnuradio 3.10 in [bkerler's fork (maint-3.10 branch)](https://github.com/bkerler/gr-gsm/tree/maint-3.10)
> 
> Ohh really ? Can I install grgsm with gnuradio 3.10 updated versions ?

I'm still using gnuradio 3.8 and I don't have experience with 3.10.  Read also the discussion in https://github.com/ptrkrysik/gr-gsm/pull/600

pixkike, 2022-07-19:
Sooo @velichkov  I wipe completely gnuradio 3.8 again and i reinstall 3.10 from kali repo and then in install gr-gsm in #600 .Building and installing worked ! But I run "grgsm_scanner" This error found  

`grgsm_scanner 
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_scanner", line 35, in <module>
    from gnuradio import gsm
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/gsm/__init__.py", line 46, in <module>
    from .gsm_python import *
ImportError: libgnuradio-gsm.so.1.0.0git: cannot open shared object file: No such file or directory
`

And I can't find gr-gsm livemon "Command not found"
like this 

`grgsm_livemon                                                                                                                                                                                                                        
grgsm_livemon: command not found`

What Ver should I install gnuradio ....

velichkov, 2022-07-20:
> i decided to delete 3.10 and remove all gnuradio and grgsm all.

Stop switching back and forth, choose a version and stick with it.

> And then reinstall gnuradio 3.8( from source) 

What parameters did you use when invoking `cmake` and `make`?

> As you said
> 
> > You need to uninstall gr-osmosdr and do a clean rebuild without libgnuradio-funcube support.
> 
> I uninstall gr-osmosdr and clean rebuild it from this [Source](https://github.com/igorauad/gr-osmosdr/tree/gr3.8) 

Better use the official osmocom repository, or it's github mirror 

https://gitea.osmocom.org/sdr/gr-osmosdr/
https://github.com/osmocom/gr-osmosdr 

> For gr 3.8 But when i run cmake in gr-osmodr I found this error
> 
> `-- Checking for module 'libfreesrp' -- No package 'libfreesrp' found -- libfreesrp not found. CMake Error at CMakeLists.txt:179 (message): GnuRadio Runtime required to build gr-osmosdr -- Configuring incomplete, errors occurred! See also "/usr/local/gr-osmosdr/build/CMakeFiles/CMakeOutput.log". See also "/usr/local/gr-osmosdr/build/CMakeFiles/CMakeError.log".`

Most probably this is a problem in the igorauad's frok. 

> So I think this is
> 
> > GnuRadio Runtime required to build gr-osmosdr
> 
> this can't link or ported with gnuradio vr 3.8
> 
> So How Do i ported or link with gnuradio 3.8 

Use the official repository.

> and How do i install this module
> 
> > -- Checking for module 'libfreesrp'
> > --   No package 'libfreesrp' found

You don't need this unless you have FreeSRP SDR device. You can install it with `sudo apt-get install libfreesrp-dev`

> this is [CMakeError.log.txt](https://github.com/ptrkrysik/gr-gsm/files/9144625/CMakeError.log.txt) [CMakeOutput.log.txt](https://github.com/ptrkrysik/gr-gsm/files/9144629/CMakeOutput.log.txt)

Ignore these two log files.

velichkov, 2022-07-20:
> I build gnuradio 3.9 with this

3.9 or 3.10

pixkike, 2022-07-20:
sorry sir 
`
gnuradio-config-info --version                                           1 ⨯
v3.8.5.0-6-g57bd109d
`
ver 3.8

pixkike, 2022-07-20:
I have downloaded osmosdr with 

>Better use the official osmocom repository, or it's github mirror
https://gitea.osmocom.org/sdr/gr-osmosdr/
https://github.com/osmocom/gr-osmosdr

when I run Cmake 

`cmake ..
-- Build type not specified: defaulting to release.
CMake Error at CMakeLists.txt:45 (find_package):
  Could not find a configuration file for package "Gnuradio" that is             
  compatible with requested version "3.9".                                       
 The following configuration files were considered but not accepted:            
  /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.5.git       
 -- Configuring incomplete, errors occurred!
See also "/usr/local/gr-osmosdr/build/CMakeFiles/CMakeOutput.log".
`

>   compatible with requested version "3.9".                                       

 Do I have to install gnuradio ver 3.9 ?

pixkike, 2022-07-20:
Dear Sir I install correspond gr-osmos from  

>https://gitea.osmocom.org/sdr/gr-osmosdr/

so I run cmake (Work Fine) and run make 

`
make                                                                                                                                                                                                                                 2 ⨯
[ 57%] Built target gnuradio-osmosdr
[ 61%] Built target _osmosdr_swig_doc_tag
[ 65%] Built target osmosdr_swig_swig_doc
[ 68%] Built target osmosdr_swig_swig_compilation
Consolidate compiler generated dependencies of target osmosdr_swig
[ 70%] Building CXX object swig/CMakeFiles/osmosdr_swig.dir/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx.o
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_source_make(PyObject*, PyObject*, PyObject*)’:
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx:16357:64: error: no match for ‘operator=’ (operand types are ‘SwigValueWrapper<std::shared_ptr<osmosdr::source> >’ and ‘osmosdr::source::sptr’ {aka ‘boost::shared_ptr<osmosdr::source>’})
16357 |       result = osmosdr::source::make((std::string const &)*arg1);
      |                                                                ^
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx:32:21: note: candidate: ‘SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const SwigValueWrapper<T>&) [with T = std::shared_ptr<osmosdr::source>]’
   32 |   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
      |                     ^~~~~~~~
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx:32:58: note:   no known conversion for argument 1 from ‘osmosdr::source::sptr’ {aka ‘boost::shared_ptr<osmosdr::source>’} to ‘const SwigValueWrapper<std::shared_ptr<osmosdr::source> >&’
   32 |   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx:36:21: note: candidate: ‘SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const T&) [with T = std::shared_ptr<osmosdr::source>]’
   36 |   SwigValueWrapper& operator=(const T& t) { SwigMovePointer tmp(new T(t)); pointer = tmp; return *this; }
      |                     ^~~~~~~~
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx:36:40: note:   no known conversion for argument 1 from ‘osmosdr::source::sptr’ {aka ‘boost::shared_ptr<osmosdr::source>’} to ‘const std::shared_ptr<osmosdr::source>&’
   36 |   SwigValueWrapper& operator=(const T& t) { SwigMovePointer tmp(new T(t)); pointer = tmp; return *this; }
      |                               ~~~~~~~~~^
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx:16367:61: error: invalid ‘static_cast’ from type ‘SwigValueWrapper<std::shared_ptr<osmosdr::source> >’ to type ‘const sptr&’ {aka ‘const boost::shared_ptr<osmosdr::source>&’}
16367 |   resultobj = SWIG_NewPointerObj((new osmosdr::source::sptr(static_cast< const osmosdr::source::sptr& >(result))), SWIGTYPE_p_std__shared_ptrT_osmosdr__source_t, SWIG_POINTER_OWN |  0 );
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx:1096:89: note: in definition of macro ‘SWIG_NewPointerObj’
 1096 | #define SWIG_NewPointerObj(ptr, type, flags)            SWIG_Python_NewPointerObj(NULL, ptr, type, flags)
      |                                                                                         ^~~
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_sink_make(PyObject*, PyObject*, PyObject*)’:
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx:18852:62: error: no match for ‘operator=’ (operand types are ‘SwigValueWrapper<std::shared_ptr<osmosdr::sink> >’ and ‘osmosdr::sink::sptr’ {aka ‘boost::shared_ptr<osmosdr::sink>’})
18852 |       result = osmosdr::sink::make((std::string const &)*arg1);
      |                                                              ^
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx:32:21: note: candidate: ‘SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const SwigValueWrapper<T>&) [with T = std::shared_ptr<osmosdr::sink>]’
   32 |   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
      |                     ^~~~~~~~
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx:32:58: note:   no known conversion for argument 1 from ‘osmosdr::sink::sptr’ {aka ‘boost::shared_ptr<osmosdr::sink>’} to ‘const SwigValueWrapper<std::shared_ptr<osmosdr::sink> >&’
   32 |   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
      |                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/usr/local/gr-osmosdr/build/swig/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx:36:21: note: candidate: ‘SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const T&) [with T = std::shared_ptr<osmosdr::sink>]’
   36 |   SwigValueWrapper& operator=(const T& t) { SwigMovePointer tmp(new T(t)); pointer = tmp; return *this; }
[... cut ...]
 1096 | #define SWIG_NewPointerObj(ptr, type, flags)            SWIG_Python_NewPointerObj(NULL, ptr, type, flags)
      |                                                                                         ^~~
make[2]: *** [swig/CMakeFiles/osmosdr_swig.dir/build.make:76: swig/CMakeFiles/osmosdr_swig.dir/CMakeFiles/osmosdr_swig.dir/osmosdr_swigPYTHON_wrap.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:532: swig/CMakeFiles/osmosdr_swig.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
`
 Found This error ...

velichkov, 2022-07-22:
> error: no match for ‘operator=’ (operand types are ‘SwigValueWrapper<std::shared_ptr<osmosdr::source> >’ and ‘osmosdr::source::sptr’ {aka ‘boost::shared_ptr<osmosdr::source>’}) 16357 | result = osmosdr::source::make((std::string const &)*arg1);

Looks like you have multiple different incompatible versions of either `gr-osmosdr` or ` gnuradio` installed at the same time, probably some leftover from your multiple back and forth switching between versions.

My recommendation is to clean everything and start from scratch. Another advice is to first test the whole installation in a virtual machine or container, this way you can easily start from scratch in a clean environment.

velichkov, 2022-07-22:
> Looks like you have multiple different incompatible versions of either `gr-osmosdr` or ` gnuradio` installed at the same time, probably some leftover from your multiple back and forth switching [...]

pixkike, 2022-07-22:
Do you mean docker I can't find docker file for grgsm

velichkov, 2022-07-22:
> Do you mean docker I can't find docker file for grgsm

`docker` is a possible option but other container technologies are suitable as well.  You can find example Dockerfiles in

pixkike, 2022-07-22:
you did not recommanded pybomb. Its that not great ....

velichkov, 2022-07-22:
> you did not recommanded pybomb. Its that not great ....

I do not recommend it. Most of the time it causes more problems then it solves.

pixkike, 2022-07-23:
hey do you have container or something .....for grgsm

velichkov, 2022-07-25:
> hey do you have container or something .....for grgsm

In one of my previous messages I gave you a link to the tests' Dockerfiles. They are in the gr-gsm's repository and you can easly build  them with
```
docker build -t gr-gsm -f tests/dockerfiles/Ubuntu_20_04.docker .
```
executed from the gr-gsm's top source directory.

pixkike, 2022-07-25:
hello sir when i run with docker i found this error 

`
Removing intermediate container 6514e3cb5299
 ---> 507f5edd3db9
Step 7/7 : RUN cmake .. &&     mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/ &&     make -j $(nproc) &&     make install &&     ldconfig &&     make CTEST_OUTPUT_ON_FAILURE=1 test
 ---> Running in 7654fc406850
CMake Error: The source directory "/src" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.                                                                                                                                                                         
The command '/bin/sh -c cmake .. &&     mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/ &&     make -j $(nproc) &&     make install &&     ldconfig &&     make CTEST_OUTPUT_ON_FAILURE=1 test' returned a non-zero code: 1
`
Please How Do i fix this

velichkov, 2022-07-26:
> CMake Error: The source directory "/src" does not appear to contain CMakeLists.txt.

In which directory did you run the `docker build ...` command and what parameters did you use?

pixkike, 2022-07-26:
Well first I make in root/dockerfiles/Dockerfile in Dockerfile i put your 
> https://github.com/ptrkrysik/gr-gsm/blob/master/tests/dockerfiles/Debian_testing.docker
file into Dockerfile and then i run
> docker build -t grgsm:Tag /root/dockerfiles
right?

velichkov, 2022-07-26:
> Well first I make in root/dockerfiles/Dockerfile in Dockerfile i put your
> 
> > https://github.com/ptrkrysik/gr-gsm/blob/master/tests/dockerfiles/Debian_testing.docker
> 
> file into Dockerfile and then i run
> 
> > docker build -t grgsm:Tag /root/dockerfiles
> 
> right?

No. Execute the command exactly as written in my previous message (don't forget the last `.`) inside the gr-gsm's top directory.

pixkike, 2022-07-26:
ohh I got it I now I will do this right away

pixkike, 2022-07-27:
Sir @velichkov  I have A one more question mine os is kali 2022 .When I find gr-gsm in kail package tracker gr-gsm is imported kali 2020 .
So what if I reset my os with kali 2020 Can I completely installed with apt install gr-gsm ?

velichkov, 2022-07-27:
> after i completely installed gr-gsm in docker idk how to open container how do i do? and i don't even have docker container

You first build an image and then run a container based on it. To open a `bash` shell in the container execute
```
docker run -it --rm gr-gsm bash
```
Depending on your SDR device and how it is connected  when executing `docker run ...` you may need to use `--network=host`, `--privileged` or `--device=/dev/ttyUSB0` options.

>  I have A one more question mine os is kali 2022 .When I find gr-gsm in kail package tracker gr-gsm is imported kali 2020 .
> So what if I reset my os with kali 2020 Can I completely installed with apt install gr-gsm ?

Sorry, I'm not Kali user, I'm not familiar with it and so I don't if this package works well or not. You can test whether it is working or not in a Virtual Machine (VM).

pixkike, 2022-07-27:
Hi Sir when I run docker with 

> docker run -it --network=host --privileged --rm gr-gsm bash 

`root@pix:/src/build# grgsm_livemon
Warning: failed to XInitThreads()
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_livemon", line 37, in <module>
    from grgsm import arfcn
ModuleNotFoundError: No module named 'grgsm'
`
I Found this error why and how to solve this

velichkov, 2022-07-29:
> ModuleNotFoundError: No module named 'grgsm' 

See https://github.com/ptrkrysik/gr-gsm/issues/480#issuecomment-532305188

velichkov, 2022-07-29:
> ModuleNotFoundError: No module named 'grgsm'

Another possible solution for Ubuntu 20.04 is to set `GR_PYTHON_DIR` to `/usr/local/lib/python3.8/dist-packages` when invoking cmake.

```diff
--- a/tests/dockerfiles/Ubuntu_20_04.docker
+++ b/tests/dockerfiles/Ubuntu_20_04.docker
@@ -22,7 +22,7 @@ COPY ./ /src/
 RUN mkdir /src/build
 WORKDIR /src/build
 
-RUN cmake .. && \
+RUN cmake -DGR_PYTHON_DIR=/usr/local/lib/python3.8/dist-packages .. && \
     # The parallel build sometimes fails when the .grc_gnuradio
     # and .gnuradio directories do not exist
     mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/ && \
```
then rebuild the image with `docker build ...`

Mhagne, 2022-08-25:
[image]

ptrkrysik, 2026-11-09:
This was reported against the old GNU Radio 3.7/3.8 (SWIG) code base. The module has been ported to GNU Radio 3.10 and no longer uses SWIG or the old build, so please retry with the new release and open a new issue with the full log if the problem persists.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik