ptrkrysik/gr-gsm #609: Configuring incomplete, errors occurred!

jihatbabuislam, 2022-07-20:
┌──(root㉿kali)-[/home/babu/gr-gsm/build]
└─# cmake ..
-- The CXX compiler identification is GNU 11.3.0
-- The C compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'gmp'
--   Found gmp, version 6.2.1
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Using GMP.
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.74.0") found components: date_time program_options system regex thread unit_test_framework 
-- Found Volk: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.10.4") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.10.so (found suitable exact version "3.10.4") 
-- Checking for module 'sndfile'
--   Found sndfile, version 1.0.31
-- Found SNDFILE: /usr/lib/x86_64-linux-gnu/libsndfile.so  
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.8
-- Found FFTW3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found Git: /usr/bin/git  
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.1") found components: doxygen dot 
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (Libosmocore).  This can
  lead to problems in calling code that expects `find_package` result
  variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.23/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  cmake/Modules/FindLibosmocore.cmake:2 (INCLUDE)
  CMakeLists.txt:132 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'libosmocore'
--   Found libosmocore, version 1.6.0
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args`
  (libosmocore) does not match the name of the calling package (Libosmocore).
  This can lead to problems in calling code that expects `find_package`
  result variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Modules/FindLibosmocore.cmake:32 (find_package_handle_standard_args)
  CMakeLists.txt:132 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found libosmocore: /usr/lib/x86_64-linux-gnu/libosmocore.so  
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (Libosmocodec).  This can
  lead to problems in calling code that expects `find_package` result
  variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.23/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  cmake/Modules/FindLibosmocodec.cmake:1 (INCLUDE)
  CMakeLists.txt:133 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'libosmocodec'
--   Found libosmocodec, version 1.6.0
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args`
  (libosmocodec) does not match the name of the calling package
  (Libosmocodec).  This can lead to problems in calling code that expects
  `find_package` result variables (e.g., `_FOUND`) to follow a certain
  pattern.
Call Stack (most recent call first):
  cmake/Modules/FindLibosmocodec.cmake:31 (find_package_handle_standard_args)
  CMakeLists.txt:133 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found libosmocodec: /usr/lib/x86_64-linux-gnu/libosmocodec.so  
-- Checking for module 'libosmocoding'
--   Found libosmocoding, version 1.6.0
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args`
  (libosmocoding) does not match the name of the calling package
  (Libosmocoding).  This can lead to problems in calling code that expects
  `find_package` result variables (e.g., `_FOUND`) to follow a certain
  pattern.
Call Stack (most recent call first):
  cmake/Modules/FindLibosmocoding.cmake:31 (find_package_handle_standard_args)
  CMakeLists.txt:134 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found libosmocoding: /usr/lib/x86_64-linux-gnu/libosmocoding.so  
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (Libosmogsm).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.23/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  cmake/Modules/FindLibosmogsm.cmake:1 (INCLUDE)
  CMakeLists.txt:135 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'libosmogsm'
--   Found libosmogsm, version 1.6.0
CMake Warning (dev) at /usr/share/cmake-3.23/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (libosmogsm)
  does not match the name of the calling package (Libosmogsm).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Modules/FindLibosmogsm.cmake:31 (find_package_handle_standard_args)
  CMakeLists.txt:135 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found libosmogsm: /usr/lib/x86_64-linux-gnu/libosmogsm.so  
-- Loading build date Wed, 20 Jul 2022 12:07:37 into constants...
-- Loading version v1.0-compat-xxx-xunknown into constants...
-- Using install prefix: /usr/local
-- Building for version: v1.0-compat-xxx-xunknown / 1.0.0git
-- No C++ unit tests... skipping
CMake Warning at docs/CMakeLists.txt:54 (message):
  rst2man from python-docutils is required to build man pages

-- Found SWIG: /usr/bin/swig4.0 (found version "4.0.2")  
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.10.so (found version "3.10.4") 
CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find requested file:

    GrSwig

CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

-- Configuring incomplete, errors occurred!
See also "/home/babu/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/home/babu/gr-gsm/build/CMakeFiles/CMakeError.log".
                                                                                   
┌──(root㉿kali)-[/home/babu/gr-gsm/build]
└─# 

please help sir

velichkov, 2022-07-20:
Hi @jihatbabuislam,

> Unknown CMake command "GR_SWIG_MAKE".

See #603, #597 and #582

Read also #600

jihatbabuislam, 2022-07-20:
sir i don't understand

jihatbabuislam, 2022-07-20:
I follow this step

git clone https://gitea.osmocom.org/sdr/gr-gsm
cd gr-gsm
mkdir build
cd build
cmake ..
mkdir $HOME/.grc_gnuradio/ $HOME/.gnuradio/
make

jihatbabuislam, 2022-07-20:
I don't know what's wrong with me

velichkov, 2022-07-20:
There is nothing wrong with you. In my first message I gave you few links to some other issues. Read all messages in these issues.

Mhagne, 2022-08-27:
CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find requested file:                                                            
                                                                                                    
    GrSwig                                                                                          
                                                                                                    
                                                                                                    
CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

Ultimator14, 2022-08-30:
This is likely an incompability between `gr-gsm` and newer `gnuradio` and/or `gr-osmosdr` versions.  
The configuration worked for me with `gnuradio-3.8.5.0-r1` and `gr-osmosdr-0.2.3`.  
This error occured with `gnuradio-3.10.3.0` and `gr-osmosdr-20210128`.