ptrkrysik/gr-gsm #613: grgsm_livemon giving issue

Lihsrah, 2022-08-10:
5.18.0-kali5-amd64

i am new to gr-gsm , sorry if i come out dumb :')

[image]

velichkov, 2022-08-11:
How did you install gnuradio and gr-gsm and which versions?

Lihsrah, 2022-08-13:
from official kali rolling repo

velichkov, 2022-08-14:
> and gr-gsm from this repo :-
> https://github.com/bkerler/gr-gsm

Then you most probably have some leftovers from a previous installation

> File "/usr/lib/python3/dist-packages/grgsm/grgsm_swig.py", line 13, in
> from . import _grgsm_swig

The version for gnuradio 3.10 no longer uses `swig` so the above file is from a previous installation that you need to remove. First remove all gr-gsm's files from `/usr/` and `/usr/local and then reinstall.

Lihsrah, 2022-08-15:
`cmake ..`   in your repo give me this 

[image]

velichkov, 2022-08-16:
> cmake .. in your repo give me this

Yes, because the `main` branch in my repo contains the version for `gnuradio 3.8`. As you have `gnuradio 3.10` you need to get the version/branch for `gnuradio 3.10`. 

Use `maint-3.10` or `maint-3.10_with_multiarfcn` branches from [bkerler's fork](https://github.com/bkerler/gr-gsm) but before installing it make sure that there are no gr-gsm's files from a previous installation in `/usr/` and `/usr/local/`.

Mhagne, 2022-08-25:
[image]

Mhagne, 2022-08-25:
in your repo give me this please help me

velichkov, 2022-08-25:
Why are you building `gr-osmosdr` from source? Why don't you install it using the package manager of your Linux distribution?

> in your repo give me this please help me

You need to install `libsndfile`. For Ubuntu/Debian/Kali run `sudo apt-get instlal libsndfile-dev`

P.S.
Do not attach screenshots of your terminal. It is much better to copy-paste the test.

Mhagne, 2022-08-25:
Sometimes it appears to me

make: *** no targets specified and no makefile found , stop

velichkov, 2022-08-25:
> make: *** no targets specified and no makefile found , stop

You either are executing `make` in a different directory then `cmake` or you have not run `cmake` before `make`

Mhagne, 2022-08-27:
CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find requested file:                                                            
                                                                                                    
    GrSwig                                                                                          
                                                                                                    
                                                                                                    
CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

Mhagne, 2022-08-27:
‏When I open {{cmake ..}}this problem appears to me
‏I deleted it more than once with the same problem, please help me

‏Error
CMake Error at swig/CMakeLists.txt:37 
(include): include could not find requested file:
GrSwig                                                                                          
CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE): Unknown CMake command "GR_SWIG_MAKE".

RFNOD, 2022-10-01:
> Yes, because the `main` branch in my repo contains the version for `gnuradio 3.8`. As you have `gnuradio 3.10` you need to get the version/branch for `gnuradio 3.10`.
>
> Use `maint-3.10` or [...]

Thanks for sharing the berkler/gr-gsm fork for GR3.10. It is compiled with no issue!!!!

ptrkrysik, 2026-11-09:
This was reported against the old GNU Radio 3.7/3.8 (SWIG) code base. The module has been ported to GNU Radio 3.10 and no longer uses SWIG or the old build, so please retry with the new release and open a new issue with the full log if the problem persists.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik