ptrkrysik/gr-gsm #614: Problem with installing gr-gsm

greyhat94, 2022-09-16:
I am install gr-gsm by following the first instructions on this page: https://osmocom.org/projects/gr-gsm/wiki/Installation
Everything is okay, but after typing the command "cmake ..", I got the following various error message saying "LIBOSMOCORE_INCLUDE_DIR-NOTFOUND" and "CMake Generate step failed.  Build files cannot be regenerated correctly". 

I have tried to type only this command "git clone https://gitea.osmocom.org/osmocom/libosmocore.git" but this didn't help. 

I have seen that gr-gsm requires also gnuradio installed and I installed by typing this command 
"sudo apt-get install gnuradio"

I'd like to know if it possible to fix all the error above mentioned and if I installed gnuradio correctly, in order to prevent getting further error. 

Thank you for your help.

tkircher, 2022-10-16:
gr-gsm has not been updated in a very long time, so it is still expecting GNU Radio to support SWIG, which it abandoned in favor of pybind. The only real solution is to use an ancient, obsolete version of GNU Radio.

ptrkrysik, 2026-11-09:
This was reported against the old GNU Radio 3.7/3.8 (SWIG) code base. The module has been ported to GNU Radio 3.10 and no longer uses SWIG or the old build, so please retry with the new release and open a new issue with the full log if the problem persists.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik