ptrkrysik/gr-gsm #615: gr-gsm on Ubuntu 22.10 (Kinetic)

MatejKovacic, 2022-10-24:
I am trying to install it, but:
- the package `gr-gsm` does not exists
- installing from source: it cannot find `python-gtk2`
- Pybombs installation first of all you need to run `sudo pip2 install PyBOMS `(pip**2**, yes!), and then it returns an `ERROR: Package has no install methd: python-zmq`

So... what is the solution?

VadimIzmalkov, 2023-02-11:
Try this one https://github.com/bkerler/gr-gsm

ptrkrysik, 2026-11-09:
This was reported against the old GNU Radio 3.7/3.8 (SWIG) code base. The module has been ported to GNU Radio 3.10 and no longer uses SWIG or the old build, so please retry with the new release and open a new issue with the full log if the problem persists.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik