ptrkrysik/gr-gsm #616: GSM NETWORK MODIFY PACKETS (KINDA LIKE BURPSUITE)

smfai200, 2022-11-03:
I'm trying to research on GSM Network and have been through IMSI Catchers and FemtoCell but i would like someone to help with the fact to send customized requests such as paging Request etc from my own parameters from a GSM module or be able to modify the Sniffed GSM Traffic to modify and replay the Request to the BTS

I've tried IMSI Catchers

gronget, 2022-11-03:
What do you mean "I have tried imsi catchers" are you referring to a
hardware or software?
What exactly is your setup?

smfai200, 2022-11-03:
That somehow has been left at the end, I meant that the available tools provide a procedure for the IMSI Capturing process whereby it stops to no further knowledge. I'm trying to be able to proxy the gsm network requests such as paging and location update and be able to modify the parameters and send it to the BTS with the altered content

smfai200, 2022-11-03:
More Like:

PHONE -> MY PROXY -> CHANGE DATA OF GSM PACKET -> FORWARD TO CELLULAR BTS
or
somehow be able to conduct the paging request from a laptop with gsm modem attached!

gronget, 2022-11-03:
You really have to understand GSM architecture and how it all works. It's
not like how you would modify packets in burpsuite

smfai200, 2022-11-03:
Can you provide a telegram to communicate

gronget, 2022-11-03:
@sishbag9

> Can you provide a telegram to communicate

@Sishbag9

ptrkrysik, 2026-11-09:
gr-gsm is a receive-only tool and does not transmit or modify network traffic, so this is out of scope for this project.

Thank you for your interest in the project.

closed 2026-11-09 by ptrkrysik