ptrkrysik/gr-gsm #618: grswig

XHXHK, 2022-12-01:
**CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find requested file:

    GrSwig

CMake Error at swig/CMakeLists.txt:51 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

-- Configuring incomplete, errors occurred!
See also "/home/zhxkhs/gr-gsm/build/CMakeFiles/CMakeOutput.log".**

XHXHK, 2022-12-01:
Why can't I install all dependencies

XHXHK, 2022-12-01:
Error will still be displayed after the installation dependency is completed

ptrkrysik, 2026-11-09:
Duplicate of #475. This was reported against the old GNU Radio 3.7/3.8 (SWIG) code base. The module has been ported to GNU Radio 3.10 and no longer uses SWIG or the old build, so please retry with the new release and open a new issue with the full log if the problem persists.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik