ptrkrysik/gr-gsm #619: gr-gsm with PlutoSDR

stevedimaras, 2022-12-13:
Hello.

Does anyone know if there is a gr-gsm version compatible with PlutoSDR ? As i tried to list the device through the arguments while running grgsm_scanner and grgsm_livemon, but keep getting the RuntimeError: Unable to find any supported SDR devices(), although iio_info -s detects it.

Thanks.

velichkov, 2022-12-16:
Hi @stevedimaras,

Yes, it should be possible to use `gr-gsm` with `PlutoSDR`.  `gr-gsm` is using `gr-osmosdr` to work with various SDR devices. it does not have direct support for `PlutoSDR` but have support for `SoapySDR` which have support for `PlutoSDR`. So first make sure that your `gr-omsosdr` is built with `SoapySDR` support and then install [SoapyPlutoSDR](https://github.com/pothosware/SoapyPlutoSDR).

stevedimaras, 2022-12-16:
Thank you for the response,

I actually followed all these steps and SoapySDRUtil --info detects my pluto device. However, when I run for example grgsm_scanner with just the -d argument I get the RuntimeError: Unable to find any supported SDR devices, and when I specify my plutoSDR device with --args, grgsm_scanner gets a Segmentation fault. I do not really understand the issue that occurs.

stevedimaras, 2022-12-16:
I actually fixed it (had two versions of soapy), it is just like you said. Thanks

closed 2022-12-16 by stevedimaras