ptrkrysik/gr-gsm #624: 'osmosdr.osmosdr_python.device_t' is not iterable Ubuntu 22.04 (Windows 10 + WSL)

VadimIzmalkov, 2023-02-21:
Hi everyone

I am using setup as follows:

- Ubuntu 22.04 (Windows 10 + WSL)
- GNURadio 3.10.1.1 (Python 3.10.6)
- GR-GSM from https://github.com/bkerler/gr-gsm (maint-3.10_with_multiarfcn branch)

When I try to capture a signal got the error:

```CLI
grgsm_capture -a34 -s1e6 test.cfile
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_capture", line 177, in <module>
    tb = gsm_capture(freq=freq,
  File "/usr/local/bin/grgsm_capture", line 57, in __init__
    str(gsm.device.get_default_args(device_args)))
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/gsm/device.py", line 53, in get_default_args
    devices = get_all_args("nofake")
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/gsm/device.py", line 43, in get_all_args
    return list(map(lambda dev: dev.to_string(), exclude(get_devices(hint))))
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/gsm/device.py", line 40, in exclude
    return [dev for dev in devices if not match(dev, filters)]
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/gsm/device.py", line 40, in <listcomp>
    return [dev for dev in devices if not match(dev, filters)]
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/gsm/device.py", line 33, in match
    if (k not in dev or dev[k] != v):
TypeError: argument of type 'osmosdr.osmosdr_python.device_t' is not iterable
```

Any suggestions?

VadimIzmalkov, 2023-02-24:
With the lat commit issue solved. 

Thanks!

closed 2023-02-24 by VadimIzmalkov

sherdil6, 2023-02-27:
Hi  @VadimIzmalkov, I am facing exact similar issue on Pluto SDR. Getting segmentation fault running grgsm_scanner as well as capture. Can you please guide me as how to resolve this? Thanks...

VadimIzmalkov, 2023-03-01:
Hi

Try to update maint-3.10_with_multiarfcn branch to last commit