ptrkrysik/gr-gsm #627: Livemon not showing Data

martinboe1980, 2023-04-12:
Hi.
I use a noelec device together with gr-gsm.
grgsm_scanner finds cell towers, and i can see the traffic in wireshark. But grgsm_livemon does not display anything after the line "allocating 15 zero-copy buffers" - i dont see the dataflow there.

What am i doing wrong?

nurzhan-zz, 2023-11-01:
Hello, you need problem?

LincolnKermit, 2023-12-21:
Please note that GR-GSM is not supported in the latest versions of Linux. Consider downgrading, or alternatively, download DragonOS where all libraries are already installed and ready to use.
I honestly suggest you Dragon OS on a laptop ( or even dual boot )
Dragon OS : [https://cemaxecuter.com/](url)

Lincoln Kermit

Stallion77RepoOfficial, 2024-01-05:
When i am changing the frequency i get no valid pll combination error. Do u know anything bout this? I am using latest version of kali.

LincolnKermit, 2024-01-05:
Go on my forks i explained, use DragonOS, livemon and grgsm are no longer maintened by his creator. Refer to my fork and use DragonOS
Sent from [Proton Mail](https://proton.me/mail/home) for iOS

Stallion77RepoOfficial, 2024-01-05:
Ok thanks can u take me to googlegroups of gr-gsm? and u sure pll error about the version of kali? i will try via DragonOS

LincolnKermit, 2024-01-06:
i did not have any errors since im using dragonos, make sure to use it on a real pc and not a virtual machine.

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

ptrkrysik, 2026-11-09:
No reproducible defect. Check gain, frequency and the ppm correction with a known strong cell first. Please open a new issue with logs if livemon still shows nothing with the new release.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.