ptrkrysik/gr-gsm #628: grgsm_capture unable to use

sam92290, 2023-05-05:
Hi all,

I'm using the distrib DragonOS Focal R2 bootable from my usb key with my RTL-SDR usb key.

livemon and gr gsm are working well. I use grgsm_scanner and it works well with volk_profile.

The pb is when I try to use the grgsm_capture.py. I've the following errors after the cmd line grgsm_capture -a 5 /tmp/capture.cfile or even grgsm_capture -f 933.4 5 /tmp/capture.cfile

[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.1.0.5-0-unknown
usb_claim_interface error -6
gr-osmosdr **0.2.0.0 (0.2.0)** gnuradio **3.10.4.0**
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya 
Using device #0 Realtek RTLXXXXUHIDIR SN: XXXXXXXX
usb_claim_interface error -6
Traceback (most recent call last):
  File "/usr/local/bin/grgsm_capture", line **177**, in <module>
    tb = gsm_capture(freq=freq,
  File "/usr/local/bin/grgsm_capture", line **56**, in __init__
    self.sdr_source = osmosdr.source(args="numchan=" + str(1) + " " +
RuntimeError: Failed to open rtlsdr device.

Someone could help to debug please ?

Thanx in advance.

++

EmptyIsNear, 2023-05-29:
Hello !
Its probably because your antenna still scanning with livemon and grgsm_scanner, so you need to stop process before start grgsm_capture.
Hope it help you :)

ptrkrysik, 2026-11-09:
As answered in the thread, the device was still in use by another process (livemon or the scanner).

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik