ptrkrysik/gr-gsm #629: error while make

Amirke1, 2023-05-25:
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/scnl/prefix-3.8/gr-gsm/python/grgsm/__init__.py", line 64, in <module>
    from .device import *
  File "/home/scnl/prefix-3.8/gr-gsm/python/grgsm/misc_utils/device.py", line 24, in <module>
    import osmosdr
  File "/home/scnl/prefix-3.8/lib/python3/dist-packages/osmosdr/__init__.py", line 25, in <module>
    from .osmosdr_swig import *
  File "/home/scnl/prefix-3.8/lib/python3/dist-packages/osmosdr/osmosdr_swig.py", line 17, in <module>
    _osmosdr_swig = swig_import_helper()
  File "/home/scnl/prefix-3.8/lib/python3/dist-packages/osmosdr/osmosdr_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_osmosdr_swig')
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_osmosdr_swig'
>>> Load Error: /home/scnl/prefix-3.8/gr-gsm/apps/grgsm_livemon.grc: Flowgraph invalid
Compilation error
make[2]: *** [apps/CMakeFiles/pygen_apps.dir/build.make:97: apps/grgsm_livemon] Error 1
make[1]: *** [CMakeFiles/Makefile2:864: apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

How can i solve this issue>?

ItsReXam, 2023-06-14:
Other answer states that using this fork allows you to solve the issue https://github.com/bkerler/gr-gsm

ptrkrysik, 2026-11-09:
This was reported against the old GNU Radio 3.7/3.8 (SWIG) code base. The module has been ported to GNU Radio 3.10 and no longer uses SWIG or the old build, so please retry with the new release and open a new issue with the full log if the problem persists.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik