ptrkrysik/gr-gsm #631: GNU Radio Block migration from ver 3.7 to 3.10 : CMake error

jds0987, 2023-08-14:
Hi

I am migrating a block previously created in GNU RADIO 3.7 to GNU Radio 3.10. However when i Run CMAKE i get the following error.

CMake Deprecation Warning at CMakeLists.txt:23 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Build type not specified: defaulting to release.
CMake Deprecation Warning at CMakeLists.txt:56 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:59 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:62 (cmake_policy):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:65 (cmake_policy):
  The OLD behavior for policy CMP0046 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.35") found components: filesystem system 
CMake Warning (dev) at /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (CppUnit).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.27/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  cmake/Modules/FindCppUnit.cmake:12 (INCLUDE)
  CMakeLists.txt:137 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (CPPUNIT)
  does not match the name of the calling package (CppUnit).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Modules/FindCppUnit.cmake:38 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:137 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Using GMP.
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.74.0") found components: date_time program_options system regex thread unit_test_framework 
CMake Warning (dev) at /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (Volk).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.27/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  cmake/Modules/FindVolk.cmake:1 (INCLUDE)
  /usr/share/cmake-3.27/Modules/CMakeFindDependencyMacro.cmake:76 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake:33 (find_dependency)
  CMakeLists.txt:145 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (VOLK) does
  not match the name of the calling package (Volk).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Modules/FindVolk.cmake:25 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  /usr/share/cmake-3.27/Modules/CMakeFindDependencyMacro.cmake:76 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake:33 (find_dependency)
  CMakeLists.txt:145 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

**CMake Error at CMakeLists.txt:145 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.**
  
  Can you help me trouble shoot this?

velichkov, 2023-08-14:
Hi @jds0987,

In my opinion the easiest approach is to start from scratch - create a new OOT module using gr_modtool, then add some c++/pytion blocks again with gr_modtool with the same name as your existing blocks, and then copy/paste the implementation from your previous version.

Also read 3.8, 3.9 and 3.10 porting guides

https://wiki.gnuradio.org/index.php?title=GNU_Radio_3.10_OOT_Module_Porting_Guide
https://wiki.gnuradio.org/index.php?title=GNU_Radio_3.9_OOT_Module_Porting_Guide
https://wiki.gnuradio.org/index.php?title=GNU_Radio_3.8_OOT_Module_Porting_Guide
 
If you have any future questions better ask them in gnuradio mailing list or IRC/Matrix channels

https://wiki.gnuradio.org/index.php/Chat
https://wiki.gnuradio.org/index.php?title=MailingLists

ptrkrysik, 2026-11-09:
This is about migrating a custom block from GNU Radio 3.7 to 3.10 and is not related to gr-gsm. The GNU Radio migration guide and mailing list are the right place.

Thank you for your interest in the project.

closed 2026-11-09 by ptrkrysik