ptrkrysik/gr-gsm #644: Error with grgsm_livemon

Nanaelie, 2024-06-01:
Hello Guys, 
I got this message when im running grgsm_livemon. please can someone help me to fix this : 

grgsm_livemon

CPU Features: SSE2+ SSE4.1+ AVX- FMA-
Using sse2 for xtrxdsp_iq16_sc32
Using sse2 for xtrxdsp_iq8_ic16
Using sse2 for xtrxdsp_iq16_ic16i
Using sse2 for xtrxdsp_iq8_ic8i
Using sse2 for xtrxdsp_sc32i_iq16
Using sse2 for xtrxdsp_iq8_sc32
Using sse2 for xtrxdsp_iq8_sc32i
Using sse2 for xtrxdsp_iq16_sc32i
Using sse2 for xtrxdsp_sc32_iq16
Using sse2 for xtrxdsp_ic16i_iq16
terminate called after throwing an instance of 'std::runtime_error'
  what():  rpcmanager: Aggregator not in use, and a rpc booter is already registered
zsh: IOT instruction (core dumped)  grgsm_livemon

velichkov, 2024-06-03:
Hi @Nanaelie,

Which Linux distribution are you using and which version? How did you install gnuradio and gr-gsm, and which versions?

> what(): rpcmanager: Aggregator not in use, and a rpc booter is already registered

Do you have an SDR device connected? Try specifying it when running the command in the `--args parameter`. For example
```bash
grgsm_livemon --args=rtl
```
Also provide the full output of 
```
grgsm_scanner --list-devices
```
```

ptrkrysik, 2026-11-09:
No reply to the request for distro, versions and --args/device list; this looks like an environment or GNU Radio setup problem. Please reopen with the requested information if it persists on the 3.10 release.

I am closing this as part of a clean-up for the new release (v0.43.0), which ports gr-gsm to GNU Radio 3.10 (plain CMake build, no SWIG, installation steps in the README). If the problem still occurs with it, please open a new issue with your operating system, the output of `gnuradio-config-info --version`, how gr-gsm and its dependencies were installed, the exact command and the complete output.

closed 2026-11-09 by ptrkrysik